mxgxw/MFRC522-python #12: CRC error

e4awais, 2015-01-20:
how to clear a crc error when it happens ? 

a soft/hard reset does not solve the problem. 

need to clear the crc error without powering up the circuit again. 

appreciate all the help. 

thanks

plushvoxel, 2015-01-20:
Tough question. It never happened to me.
Here on page 29 is a section on the CRC processor http://www.nxp.com/documents/data_sheet/MFRC522.pdf

e4awais, 2015-01-20:
thanks, 

but when i flush the fifo buffer, even if i re-initialize, the chip simply does not respond. 

any ideas ? 

thanks

plushvoxel, 2015-01-20:
Does it solely happen when you use this library?
Maybe you could try using an Arduino or any other platform to test the functionality of the reader.

e4awais, 2015-01-20:
i m using another library posted on github, i posted my question there as well. 

my controller is a 32 but arm. 

the code works fine except the crc error.