mxgxw/MFRC522-python #18: MFRC522 not read

Jakkawad, 2015-11-30:
i use RPi jessie and MFRC522, setup install python-dev , SPI-Py, MFRC522-python
i use RFID tag it show me
[image]
  
what i miss

thank you

Liriel, 2015-12-05:
same here!
I'm pretty sure I connected the GPIO pins correctly (checked twice), enabled SPI and rebooted the pi.
got two SPI devices:
```bash
root@raspberrypi:/home/pi# ls /dev/spi*
/dev/spidev0.0  /dev/spidev0.1
```
using **raspian jessi w/ kernel 4.1.13**

Liriel, 2015-12-05:
haha - realized that I had connected the MFRC522 to the RPI2 using a pic of RP1
wasted two hours but now it works fine

Jakkawad, 2015-12-06:
thank you i will upgrade kernel

Jakkawad, 2015-12-07:
i upgrade kernel but it now work?

buzai, 2016-04-21:
I use pi3，how can i connect？