mxgxw/MFRC522-python #20: Port to JavaScript available ...

nkolban, 2016-01-20:
I needed to be able to read MFRC522 reader on a Pi from JavaScript (Node.js).  What I did was port the code from Python to JavaScript.   I'd like to offer this up to you as a new repository.  If you are not interested in taking the code, let me know and I'll create my own ... the reason I want to offer it up to you is that I don't know how the MFRC522 works ... and merely ported code ... the JavaScript code works fine but I feel that the ownership of such an offering is best held by someone who owns MFRC522 knowledge at the core.

kolban1@kolban.com
[mfrc522.js.txt](https://github.com/mxgxw/MFRC522-python/files/98208/mfrc522.js.txt)

fantasywind, 2016-01-20:
feel good about js porting :+1:

i12maroa, 2019-01-27:
Could I use that for read RFID  from Django web server in the client side?