mxgxw/MFRC522-python #22: How to Change rst pin ?

darshkpatel, 2016-02-08:
Hey, I wanted to add a RFID reader in an Existing project , but GPIO - 25 is already in use . So how can i change the reset pin in the code ?

boconn7782, 2016-03-19:
I am trying to do the same thing. I was looking through MFRC522.py and it looks like the reset pin is set on line 10 ( NRSTPD=22). I am guessing you can change that value. I'm just researching parts right now and looking into the complexity so I haven't had a chance to test this yet. I will when all parts come in though.

darshkpatel, 2016-03-19:
@boconn7782 , I just wanted to read an RFID tag so i simply left the RST pin unplugged, and hopefully it worked :smile: !
You can try the same

closed 2016-03-19 by darshkpatel