mxgxw/MFRC522-python #24: Authentication Error with NTAG216

domschiener, 2016-04-05:
Was able to setup the RC522 and get it running with the card and tag that came shipped with it. I tried reading my other NFC sticker tags (NTAG216), and I keep getting authentication errors:

```
AUTH ERROR!!
AUTH ERROR(status2reg & 0x08) != 0
Authentication error
```

Anyone have an idea on how to fix this?

Cotix, 2016-05-19:
I wrote that code a pretty long time ago, and it is pretty much what it says it is.. An authentication error. I am not familiar with NTAG216, but either your supplied key is wrong(you need to supply a key to access a block, default key is [0xFF]*6, but this might differ on your chip), or NTAG216 does not use the same MFRC522 protocol.

Liriel, 2017-03-24:
came across this issue with NDEF tags
there are a bunch of [Default Keys](https://learn.adafruit.com/adafruit-pn532-rfid-nfc/mifare#note-on-authentication) that you should try...

found mine by trying each key for sectors 3 - 15 :)

TheRealBanana, 2017-04-22:
Also having the same issue. I've tried all the common auth keys as well as all the different forks out there of this library with no success. I have at least verified that the MFRC522 can for sure read the NTAG213 and NTAG216 rfid tags using the arduino MRFC522 library. Maybe comparing the way that library works vs this one would help.

soporteHW, 2017-09-18:
Also having same issue. I use this library: https://stm32f4-discovery.net/2014/07/library-23-read-rfid-tag-mfrc522-stm32f4xx-devices/

When I try to authenticate my mifare card using MFRC422_Auth(*) functions, It returns error -> the MFRC522_REG_STATUS2 is always 0 value.
In my case, I want to read bolck 3, sector 16 with key FFFFFFFF. The call function is as follows:

**_statusRFID = TM_MFRC522_Auth(0x60,63, RFID_KEY, CardID)_**

This returns statusRFID = MI_OK; but later when I call _**readRegister(MFRC522_REG_STATUS2) & 0x08)**_ , statusRFID turns MI_ERR.

Why?