mxgxw/MFRC522-python #29: Issues with the latest version of Raspbian jessie

danbuntu, 2016-06-27:
The newer version of Raspbian have switched to using dttree when enabling SPI. When you do this upgrade this library stops working. Running the Read.py example just sits there.

EECSdude, 2016-06-28:
Activity here: https://github.com/raspberrypi/linux/issues/1547

but no solution yet.

Jbithell, 2016-06-30:
+1 - Having huge issues with the latest release and SPI

mab5vot9us9a, 2016-07-08:
If you take a look at [Issue #1547](https://github.com/raspberrypi/linux/issues/1547), there is a workaround now!

When cloning the SPI-Py library, changing '.cs_change = 1' to '.cs_change = 0' (line 189 in spi.c) before running `sudo python setup.py install` fixes the issues I had. Alternatively you can also clone my fork of this library [https://github.com/mab5vot9us9a/SPI-Py](https://github.com/mab5vot9us9a/SPI-Py), so you don't have to make the changes yourself.

Also make sure you have the latest kernel (>= ```4.4.14```), because it fixes a bug in earlier versions (see  [Issue #1547](https://github.com/raspberrypi/linux/issues/1547#issuecomment-230170202)).
You can check your version using:
```
uname -a | sed -E "s/.*([0-9]+\.[0-9]+\.[0-9]+)-v.*/\1/"
```

If you need to update your kernel, use ```sudo rpi-update```. If it's not yet installed, do so using ```sudo apt-get install rpi-update```.

thijstriemstra, 2016-09-09:
@mab5vot9us9a can you make a pull request?