mxgxw/MFRC522-python #30: Adapt Antenna Gain

tdullers, 2016-07-23:
Hi 

Any clue on how to adapt the receiver gain in your example ?

espenoh, 2016-07-23:
Not sure what you mean by "adapt", but you can control the antenna gain using the _RFCfgReg_, as documented in the [datasheet](http://www.nxp.com/documents/data_sheet/MFRC522.pdf) (9.3.3.6, page 59).

Something like this in `MFRC522_Init`:
`self.Write_MFRC522(self.RFCfgReg, (0x07<<4))`
should give you 48 db (max) gain.

tdullers, 2016-07-24:
Hi, 

yes that is exactly what I meant ! 
So, is it as simple as editing MFRC522.py in your example and adding this line ?

  def MFRC522_Init(self):
    GPIO.output(self.NRSTPD, 1)
    **self.Write_MFRC522(self.RFCfgReg, (0x07<<4))**

espenoh, 2016-07-24:
Yeah that should work. 

I did some testing with this a long time ago, with little notable effect. Though in retrospect I think that was due to the antenna being mounted a bit too close to the ground plane of a PCB I had in the same enclosure (which definitely worsened the reception). Feel free to report back if you're seeing any noticeable change.

tdullers, 2016-07-24:
OK, I need to do some more controlled testing but some first tests showed that with antenna set to maximum gain I could a small bracelet tag to read reliably with a 1 cm acrylplate in front of the rfid reader, without even touching the plate, with normal gain I could only get that to work with a 2-3 mm plate. With a normal rfid card tag I even got reliable signals with a 4 cm acryl plate above the reader.
Will let you know when I have some more controlled results. However the RC522 seems a very solid reader, for sure compared to some commercial solutions that do often cost a few 100 usd,....

Liriel, 2017-04-06:
totally works and increases the distance significantly from which the tags can be read!

guleus, 2018-07-19:
Hello,
just somehow got to your cool module so i thought i ask you.
I bought some really cheap MFRC-522 reader/writers for my raspberry pi RFID project.
Now i started testing the things, but was really disappointed by the distance of reading the passive tags (<1 cm)
What i want to do is to achieve reading tags with a raspberry in a distance, lets say 0.5-1 meter and through clothes
Do I need to get an antenna a differenet reader? Any good suggestion. How far does your reader read and trhorugh clothes too??
Thanks a lot

Liriel, 2018-07-19:
Hi,

 

I am afraid I don’t have had a very good experience with the module either. It seems that small or bent tags are especially hard to read and not very well supported by the module.

What works well are Cards and those keychain tags that are given away with the tag. I can read them from like 3cm or so.

You probably need a completely different hardware (antenna and tags) for your requirement, like the anti theft gates you see in a supermarket.

 

HTH :)

ThomasWoeg, 2018-09-24:
> Hi,

yes that is exactly what I meant !
So, is it as simple as editing MFRC522.py in your example and adding this line ?

def MFRC522_Init(self):
GPIO.output(self.NRSTPD, 1)
self.Write_MFRC522(self.RFCfgReg, (0x07<<4))

Doing this didn't actually change anything for me

himanipangasa, 2019-03-07:
how to increase the read range of MFRC522 in raspberry pi 3?
please help