mxgxw/MFRC522-python #40: Code not working with Raspberry pi model B+

mohamedbelhassen, 2017-04-23:
I would like to note that this code did not worked with the model B+
when executing (sudo python Read.py) I just got the following message:

Welcome to the MFRC522 data read example
Press Ctrl-C to stop.

when I touch the tag nothing happens.
Is there any diagnostic steps that I can follow to discover where is the bug?

jozobozo301, 2017-05-23:
First check wires:

| Name | Pin # | Pin name   |
|------|-------|------------|
| SDA  | 24    | GPIO8      |
| SCK  | 23    | GPIO11     |
| MOSI | 19    | GPIO10     |
| MISO | 21    | GPIO9      |
| IRQ  | None  | None       |
| GND  | Any   | Any Ground |
| RST  | 22    | GPIO25     |
| 3.3V | 1     | 3V3        |

If this is correct go to any tutorial and follow steps to install python-dev, enable SPI and change other settings on raspberry. If you are still stuck try to take fresh image and go by this tutorial; http://www.instructables.com/id/Raspberry-Pi-3-Model-B-MIFARE-RC522-RFID-Tag-Readi/

koddr, 2018-04-22:
Similar problem. All wires are correct.

dingdayu, 2018-07-22:
Similar problem. 
Board D1 indicator light,No response when approaching the card.

```
pi@raspberrypi:~/Dev/MFRC522-python $ lsmod | grep spi
spidev                 16384  0
spi_bcm2835            16384  0
```

```
pi@raspberrypi:~/Dev/MFRC522-python $ sudo apt-get install python2.7-dev
正在读取软件包列表... 完成
正在分析软件包的依赖关系树
正在读取状态信息... 完成
python2.7-dev 已经是最新版 (2.7.13-2+deb9u2)。
python2.7-dev 已设置为手动安装。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 1 个软件包未被升级。
```

TJMurph, 2018-09-20:
I've got the same issues as the above - same reading as dingdayu. Followed the directions at http://www.instructables.com/id/Raspberry-Pi-3-Model-B-MIFARE-RC522-RFID-Tag-Readi/
No errors or anything - just doesn't read.

I've actually got two Raspberry pi's with the same problem.  One a model B V1.2 (in black case in photos 1 and 2) and one a B+ 2017 (in clear case photos 3 and 4).  They are wired identically and using the same colour scheme - I can't see any problem with the wiring and I've tried replacing the wires.
[image]
[image]
[image]
[image]

> $ lsmod | grep spi
> spidev                 16384  0
> spi_bcm2835            16384  0

Its powering up alright. I'm getting the following:

> $ ls /dev/spi*
> /dev/spidev0.0   /dev/spidev0.1

The config.txt is showing:

> dtparam=spi=on

When I run lsmod I see:

> $ lsmod
> Module                  Size  Used by
> fuse                  110592  3
> rfcomm                 49152  4
> bnep                   20480  2
> hci_uart               36864  1
> btbcm                  16384  1 hci_uart
> serdev                 20480  1 hci_uart
> bluetooth             368640  29 hci_uart,bnep,btbcm,rfcomm
> ecdh_generic           28672  1 bluetooth
> brcmfmac              307200  0
> brcmutil               16384  1 brcmfmac
> cfg80211              573440  1 brcmfmac
> spidev                 16384  0
> rfkill                 28672  6 bluetooth,cfg80211
> snd_bcm2835            32768  1
> snd_pcm                98304  1 snd_bcm2835
> snd_timer              32768  1 snd_pcm
> snd                    69632  5 snd_timer,snd_bcm2835,snd_pcm
> spi_bcm2835            16384  0
> fixed                  16384  0
> uio_pdrv_genirq        16384  0
> uio                    20480  1 uio_pdrv_genirq
> evdev                  24576  6
> i2c_dev                16384  0
> ip_tables              24576  0
> x_tables               32768  1 ip_tables
> ipv6                  434176  22

I'm at my wits end trying to get this to work now for an office access card system.  Can I please have help?

TJMurph, 2018-09-20:
Sorry should also add to the above that I've also tried two different RC522's.  Currently got one hooked up to each pi but both having the same problem described at top of thread.  All seems installed fine but not actually reading.

ThomasWoeg, 2018-09-24:
Im not an expert, but the tutorial I followed suggested, that you have to enable bcm2708 instead of bcm 2835. I did that by adding the following line to the end of /boot/config.txt
`dtoverlay=spi-bcm2708`
Maybe you can try this, I hope this helps!

TJMurph, 2018-09-24:
Thought that just related to the silicone of the chip itself which is different in the pi 3B.  Since I'm out of ideas I gave it a go anyway though and still not working.  Thanks for helping though.

ThomasWoeg, 2018-09-24:
Well I bought a Raspberry Pi 3B+ two weeks ago and got it working following this git
[https://github.com/mxgxw/MFRC522-python](url)
So I'm not sure if your problem has something to do with it being a newer Pi

ThomasWoeg, 2018-09-24:
I'm assuming you tried to run the Read.py file. What did that show?

TJMurph, 2018-09-24:
Same as the initial post unfortunately:

> Welcome to the MFRC522 data read example
> Press Ctrl-C to stop.
> 
> when I touch the tag nothing happens.

TJMurph, 2018-09-24:
I did have the same problem with the 3B (two pi's tried and 2 different RFID readers)

TJMurph, 2018-09-24:
Would you possibly be able to post the read outs similar to the above you are getting with a working version please?  Maybe I can work it out from there.

ThomasWoeg, 2018-09-24:
Have you tried debugging or putting prints in the MFRC library, to see exactly which step, seems to be not working?

TJMurph, 2018-09-24:
Didn't hit any problems along the way when installing.  Just didn't work at the end.

ThomasWoeg, 2018-09-24:
> Would you possibly be able to post the read outs similar to the above you are getting with a working version please? Maybe I can work it out from there.

Im not sure that I understand what you want from me, but this is the output I get (when it's working)

`pi@raspberrypi:~/MFRC522-python $ sudo python Read.py
Welcome to the MFRC522 data read example
Press Ctrl-C to stop.
Card detected
Card read UID: 69,245,238,117
Size: 8
Sector 8 [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]`

TJMurph, 2018-09-24:
Could you tell me how things look when you run lsmod?

TJMurph, 2018-09-24:
and confirm that you are using the 
Stretch Version: 2.8.2
Release date: 2018-06-27

ThomasWoeg, 2018-09-24:
my lsmod output
`pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
rfcomm                 49152  6
bnep                   20480  2
hci_uart               36864  1
btbcm                  16384  1 hci_uart
serdev                 20480  1 hci_uart
bluetooth             368640  29 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           28672  1 bluetooth
fuse                  106496  3
binfmt_misc            20480  1
evdev                  24576  6
joydev                 20480  0
spidev                 16384  0
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              573440  1 brcmfmac
rfkill                 28672  6 bluetooth,cfg80211
snd_bcm2835            32768  1
snd_pcm                98304  1 snd_bcm2835
snd_timer              32768  1 snd_pcm
snd                    69632  5 snd_timer,snd_bcm2835,snd_pcm
spi_bcm2835            16384  0
uio_pdrv_genirq        16384  0
fixed                  16384  0
uio                    20480  1 uio_pdrv_genirq
i2c_dev                16384  0
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  434176  22
`
how do i lookup the version?

TJMurph, 2018-09-24:
If you bought it two weeks ago and downloaded the latest image rather than an old one it’ll be that

TJMurph, 2018-09-24:
Only lines I can see that differ are:
fuse 110592 3	        fuse 106496 3
rfcomm 49152 4	rfcomm 49152 6
...then you have the following that I don't at all:
binfmt_misc 20480 1
joydev 20480 0

Unfortunately I lack the skills to know any more about why yours works and mine doesn't.

ThomasWoeg, 2018-09-24:
What type of RFID tags do you have?

I noticed that MIFARE DESfire EV1 dont work, but EV2 do work

TJMurph, 2018-09-24:
I appreaciate the help anyway.  Seems unlikely to be hardware problems on two different pi's with two different RC522's.

Erjona1, 2019-05-23:
Hello,
has anyone been able to solve this?
I am having the exact same issue.
I installed successfully but i cant make it read :/

mohamedbelhassen, 2019-05-23:
Hello,
In my case, the problem was the RFID board itself. I just used another one
from another seller and it worked fine

obdbeats, 2019-08-08:
I was facing the same problem with MFRC522 and i just solve it connecting the RC522 IRQ pin into PIN 18 of GPIO

deatheater01, 2020-02-04:
Still not working.
@obdbeats

obdbeats, 2020-02-17:
> Still not working.
> @obdbeats

Yes! It works for me! ;) Keep trying Bro!

alexanderzirl, 2021-04-21:
Does not work for me. Any other ideas? Any test methods I could use?