mxgxw/MFRC522-python #52: Problems by Starting the program

luismauriz1, 2018-03-24:
Hello,
 i have a small problem with my MFRC522.
All the pins are connected correctly. but now, when i want to start the programm the pi shows me 
`python Read.py
Traceback (most recent call last):
  File "Read.py", line 41, in <module>
    MIFAREReader = MFRC522.MFRC522()
  File "/home/pi/Desktop/MFRC522-python/MFRC522.py", line 133, in __init__
    GPIO.setup(NRSTPD, GPIO.OUT)
NameError: global name 'NRSTPD' is not defined
`

I hope that you can help me.
Thanks a lot for your help!

ItZNoss, 2018-03-25:
I'm having the same issue, I've been searching for hours with little luck all I know if from reading the error, if you go into that file and type nano MFRC522.py and go to the 133rd line it says GPIO.setup(NRSTPD, GPIO.OUT) thats all I've learned I'll keep searching and if I find a solution I'll share it with you :)

Jbithell, 2018-03-25:
Are you guys using latest from this lib - or some kind of fork?

Jbithell, 2018-03-25:
Part of me wonders if its related to this - could you try changing this line back?
https://github.com/mxgxw/MFRC522-python/pull/51/commits/68963702c1cc9718c0fc3a6c666cd3e1d65adaa5

luismauriz1, 2018-03-25:
i use the latest version from this lib. i will try the changing of values now. Thanks!

luismauriz1, 2018-03-25:
I changed line 133 NRSTPD to 22. Now, its working perfektly. TY

mxgxw, 2018-03-26:
Hi All! Please check if it's working now. In a previous merge there was a reference to the global NRSTPD (reset line) that is actually defined inside the class. It should be working now. Please confirm if it's working right for you to close this issue.