mxgxw/MFRC522-python #53: Limiting from Continuous Scanning

mprythero, 2018-04-14:
I was curious if there was a way to limit the read script from a continuous scan to limit a scan recognition to every five seconds or even a minute. I've just noticed that if I hover an RFID card over the sensor that it will continue to read until I swipe it away. And sometimes if it's a fast swipe, it could still read two or three times. 

Is there a way to fix this or does anyone have any suggestions? Thanks~ Matt

Jbithell, 2018-04-15:
It sounds like you are more after scanning continually, but not reading a specific card more than once per a certain time interval, instead of actually stopping/starting a scan?

cscuilla, 2020-10-08:
make it sleep for a second after detecting a card.. time.sleep(1)