mxgxw/MFRC522-python #56: Gain

guleus, 2018-07-19:
Hello,
just somehow got to your cool module so i thought i ask you.
I bought some really cheap MFRC-522 reader/writers for my raspberry pi RFID project.
Now i started testing the things, but was really disappointed by the distance of reading the passive tags (<1 cm)
What i want to do is to achieve reading tags with a raspberry in a distance, lets say 0.5-1 meter and through clothes
Do I need to get an antenna a differenet reader? Any good suggestion. How far does your reader read and trhorugh clothes too??
Thanks a lot

mxgxw, 2018-07-19:
Hi guleus! This type of RFID readers are designed for NFC technology. NFC uses magnetic coupling between two coils (antennas) to transfer information between two devices. This allows to reduce the size of the antenna but limits the distance of the reader. Some people has been able to design pretty big antennas able to read NFC cards at several centimeters but this kind of cards are not the most adequate technology if you want to read tags at longer distances.

You could use instead UHF RFID tags with a long-range antenna. A reader like this one: [SparkFun Simultaneous RFID Reader ](https://www.sparkfun.com/products/14066) + this antenna: [UHF RFID Antenna (TNC)](https://www.sparkfun.com/products/14131) + this tags [UHF RFID tags](https://www.sparkfun.com/products/14151) could work. But... Obviously this option is not cheap anymore.

guleus, 2018-07-20:
Hi mxgxw,
Thanks for the heads up!! Do you know anybody who implemented the Sprkfun reader?

8TAnyx, 2018-09-04:
Hej,

I use an RC522 reader, wouldn't it be possible to enhance the reach at least so the signal would pass some plastic box?
Thanks in advance!
cheers Daniel

8TAnyx, 2018-09-04:
Software sided of course =)