mxgxw/MFRC522-python #57: Read.py Error

JoshuaFurman, 2018-07-27:
I've followed this tutorial (https://www.piddlerintheroot.com/rfid-rc522-raspberry-pi/) to the T, and I keep getting the error " Traceback (most recent call last):
  File "Read.py", line 41, in <module>
    MIFAREReader = MFRC522.MFRC522()
  File "/home/pi/MFRC522-python/MFRC522.py", line 131, in __init__
    spi.openSPI(device=dev,speed=spd)
AttributeError: 'module' object has no attribute 'openSPI' "
whenever I try to run the Read.py file. 
Any help?

huhn511, 2019-03-24:
Hey, this trick fixed it for me:
[https://stackoverflow.com/a/55024642](https://stackoverflow.com/a/55024642)

MilesWK, 2023-03-09:
@huhn511 I have the same problem as @JoshuaFurman and that didn't fix it for me. @JoshuaFurman I feel the frustration and and will get back to you when I get more news.