mxgxw/MFRC522-python #66: Maintained?

mbhatt1, 2019-01-05:
Hi, 

Is this repo being maintained? I am having trouble using it for Jessie. Help? Connections are correct. The kernel module for spi is loaded. The Read.py file does nothing however.

RagingTyrant, 2019-01-08:
No.  Try https://github.com/ondryaso/pi-rc522  It seems to be more supported, though I'm just getting started myself.

mxgxw, 2019-01-08:
Hi there!

This repo is no longer mantained. However I keep it open because many people uses it as an implementation reference. I think this was the first port of the code for this reader to the Raspberry Pi and many people stills uses it for of basic things.

Later I will update the README with references to other more updated implementations.

Regards,
Mario.

mbhatt1, 2019-01-17:
I got it working. Had to make some changes

Thanks!

closed 2019-01-17 by mbhatt1