mxgxw/MFRC522-python #68: problem with the Write.py script.

steaward, 2019-02-20:
When trying to run the Write.py script, I get the following error: 

`Traceback (most recent call last):`
  `File "Write.py", line 6, in <module>`
   ` reader = SimpleMFRC522.SimpleMFRC52`2()
 ` File "/home/pi/MFRC522-python/SimpleMFRC522.py", line 14, in __init__`
   ` self.READER = MFRC522.MFRC522()  `
`  File "/home/pi/MFRC522-python/MFRC522.py", line 117, in __init__  `
   ` self.MFRC522_Init()  `
`  File "/home/pi/MFRC522-python/MFRC522.py", line 390, in MFRC522_Init  `
`    self.MFRC522_Reset();  `
`  File "/home/pi/MFRC522-python/MFRC522.py", line 120, in MFRC522_Reset  `
`    self.Write_MFRC522(self.CommandReg, self.PCD_RESETPHASE)  `
`  File "/home/pi/MFRC522-python/MFRC522.py", line 123, in Write_MFRC522  `
`    spi.transfer(((addr<<1)&0x7E,val))  `
`TypeError: function takes exactly 2 arguments (1 given)`

Using **Raspbian Stretch** on a Pi 3.

BombayCoast, 2019-02-21:
Same problem here. Doing it on a Raspberry pi 3 b+ with raspbian stretch.

te-ti, 2019-02-21:
Same problem with python 2.7.13 on Raspbian 9.

sandeeppusthey, 2019-02-21:
Me too having the same issue when Read.py is initialised

steaward, 2019-02-21:
@BombayCoast @te-ti @sandeeppusthey 

Instead of going in and fixing the function call, I found it much easier to use [pi-rc522]( https://github.com/ondryaso/pi-rc522).

Works no problem with Python 2.7, and I have tested it using the MFRC522 chip. 
Takes a bit of reading, however. 

Cheers!

aa003, 2019-02-22:
Same problem with write.py on Raspberry Pi 3 Model B with python 3.5.3

abrakhim, 2019-02-24:
Check my comment for issue #69 for a fix..