mxgxw/MFRC522-python #69: facing thise error when am trying to run the Read.py

sandeeppusthey, 2019-02-21:
below is the stack trace for the error am facing when am trying to run the Read.py script..

Traceback (most recent call last):
  File "Read.py", line 41, in <module>
    MIFAREReader = MFRC522.MFRC522()
  File "/home/pi/MFRC522-python/MFRC522.py", line 135, in __init__
    self.MFRC522_Init()
  File "/home/pi/MFRC522-python/MFRC522.py", line 406, in MFRC522_Init
    self.MFRC522_Reset();
  File "/home/pi/MFRC522-python/MFRC522.py", line 138, in MFRC522_Reset
    self.Write_MFRC522(self.CommandReg, self.PCD_RESETPHASE)
  File "/home/pi/MFRC522-python/MFRC522.py", line 141, in Write_MFRC522
    spi.transfer(((addr<<1)&0x7E,val))
TypeError: function takes exactly 2 arguments (1 given)

te-ti, 2019-02-21:
I'm facing same issue with python 2.713

sandeeppusthey, 2019-02-21:
Hi, Have you tried with 3.5!

te-ti, 2019-02-21:
Yes, it's seems that it's not compatible with python3 as it require parenthesis for `print`command.

sandeeppusthey, 2019-02-21:
you can add parenthesis for all prints and try run.. i just tried on 2.7 as well same error for me.. am stuck here at this part

sandeeppusthey, 2019-02-21:
i hope it is not hardware problem. coz mine is new

te-ti, 2019-02-21:
The hardware is working. I have the problem after reinstalling raspbian.

sandeeppusthey, 2019-02-21:
@te-ti oh you were able to run it previously?

te-ti, 2019-02-21:
I've figure out the problem I guess. I think it come from https://github.com/lthiery/SPI-Py library.

sandeeppusthey, 2019-02-21:
did you find anything? please share thanks if is working for you

sandeeppusthey, 2019-02-21:
this is the snapshot of the locals when am trying to debug
[image]

Highlife93, 2019-02-21:
same problem here it seems there are some problems with the recent updates from SPI-Py

te-ti, 2019-02-21:
If you "rollback" SPI-Py to commit before February 17th, it should work.
```
git clone https://github.com/lthiery/SPI-Py.git
git checkout 8cce26b9ee6e69eb041e9d5665944b88688fca68
sudo python setup.py install
```

sandeeppusthey, 2019-02-21:
Error is fixed but my tag not detected..

te-ti, 2019-02-21:
Same, I don't understand why.

DreamContribution, 2019-02-22:
@te-ti emmm...I just want know have you fixed the problem?
Change another one?Or fix the code?
It's very important to me.
Thank you very much!

sandeeppusthey, 2019-02-22:
let me know if this works anyone.. thanks

sandeeppusthey, 2019-02-22:
> If you "rollback" SPI-Py to commit before February 17th, it should work.
> 
> ```
> git clone https://github.com/lthiery/SPI-Py.git
> git checkout 8cce26b9ee6e69eb041e9d5665944b88688fca68
> sudo python setup.py install
> ```

atlast working for me.. i have soldered the reader pins then it works..

te-ti, 2019-02-22:
@sandeeppusthey Still not working for me... Can you post a picture of how you wire the pins from rfid reader to the pi board?

sandeeppusthey, 2019-02-22:
[image]

the pins must be tightly fixed into the board..

abrakhim, 2019-02-22:
Hi all, 
i stumbled on this as well on a RPI 3 B+ and Raspian Strech. First i changed GPIO 22 to 11, because of this comment "SPI_CLK on a RPi 3 is Pin number 23 = GPIO11 (not 22)" in 
https://www.raspberrypi.org/forums/viewtopic.php?t=147291
Then i got this error. Not much of a python pro, but figured it out by looking in the SPI test_script:
https://github.com/lthiery/SPI-Py/blob/master/test_script.py
The changed function spi.transfer wants the file descriptor from spi.openSPI. I saved it in a global var and added it where needed.
So, by making these changes to MFRC522.py, it will run again with latest SPI.
After downloading this file, rename to MFRC522.py, then overwrite your file in cloned git.
With git diff MFRC522.py you can see the changes made.

[MFRC522.py.txt](https://github.com/mxgxw/MFRC522-python/files/2896078/MFRC522.py.txt)

DreamContribution, 2019-02-23:
@abrakhim Cool！It works！Oh ,mine is Raspberry 2B.So not the pin problem.
[image]
Thank you very much!
@te-ti And I'm was not forget your advise!Without your's I cannot fix it so quickly!

feiyang3x, 2019-03-10:
英语很差,用中文回答吧.因为新提交的SPI-Py中,spi.transfer需要两个参数,在MFRC522.py中定义device_0=spi.openSPI(device="/dev/spidev0.0", mode=0, speed=1000000),修改 spi.transfer( ((addr<<1)&0x7E,val))和val = spi.transfer((((addr<<1)&0x7E) | 0x80,0))为spi.transfer(self.device_0, ((addr<<1)&0x7E,val))和val = spi.transfer(self.device_0, (((addr<<1)&0x7E) | 0x80,0))即可.

rubenoosterwijk, 2019-03-11:
> > If you "rollback" SPI-Py to commit before February 17th, it should work.
> > ```
> > git clone https://github.com/lthiery/SPI-Py.git
> > git checkout 8cce26b9ee6e69eb041e9d5665944b88688fca68
> > sudo python setup.py install
> > ```
> 
> atlast working for me.. i have soldered the reader pins then it work

@te-ti  i tried this : 
git clone https://github.com/lthiery/SPI-Py.git
git checkout 8cce26b9ee6e69eb041e9d5665944b88688fca68
sudo python setup.py install

after git checkout .... i got this:
fatal: Not a git repository (or any of the parent directories): .git
what have i done wrong?

MatsCedergren, 2019-03-11:
> @te-ti i tried this :
> git clone https://github.com/lthiery/SPI-Py.git
> git checkout 8cce26b9ee6e69eb041e9d5665944b88688fca68
> sudo python setup.py install
>
> after git checkout .... i got this:
> fatal: [...]

Have you "cd" into the "SPI-Py" directory before running git checkout?

dbsfai, 2019-03-12:
just copy the MFRC522.py from the following project and it works.
git clone https://github.com/pimylifeup/MFRC522-python.git

rubenoosterwijk, 2019-03-18:
@MatsCedergren 
Yes i did, still the same:
pi@raspberrypi:~ $ git checkout 8cce26b9ee6e69eb041e9d5665944b88688fca68
fatal: Not a git repository (or any of the parent directories): .git

Panjiajin, 2019-03-29:
> me, I don't understand wh

it is the issue of [ spi.transfer() ],it need two arguments and the MFRC522.py file just one given!

JoinBelieve, 2019-05-10:
it need install python2?

alexball80, 2019-05-15:
I can now detect card on Read.py. 
My problem was that I use python3 on Raspberry Pi3.
I do some change:
- after downloaded, sudo python3 setup.py install on /SPI-py (latest version)
- Then i've changed all print in MFRC522.py like this print "test" -> print("test")
- The NRSTPD is set to 11 in MFRC522.py

SciMad, 2019-07-16:
> 8cce26b9ee6e69eb041e9d5665944b88688fca68

Did you cd to the project directory?

diegohoyola, 2019-09-29:
I've could resolve the issue after using older branch.
First you clone it after that you enter into the recently created folder.
and after that you execute this:
git clone https://github.com/lthiery/SPI-Py.git
git checkout -b 8cce26b9ee6e69eb041e9d5665944b88688fca68
sudo python setup.py install

this worked as a charm for me hopefully it helps others.
Thank you.

keitheyre, 2019-10-17:
> Hi all,
> i stumbled on this as well on a RPI 3 B+ and Raspian Strech. First i changed GPIO 22 to 11, because of this comment "SPI_CLK on a RPi 3 is Pin number 23 = GPIO11 (not 22)" in
> https://www.raspberrypi.org/forums/viewtopic.php?t=147291
> Then i got this error. Not much of a python pro, but figured it out by looking in the SPI test_script:
> https://github.com/lthiery/SPI-Py/blob/master/test_script.py
> The changed function spi.transfer wants the file descriptor from spi.openSPI. I saved it in a global var and added it where needed.
> So, by making these changes to MFRC522.py, it will run again with latest SPI.
> After downloading this file, rename to MFRC522.py, then overwrite your file in cloned git.
> With git diff MFRC522.py you can see the changes made.
> 
> I u are not on a RPi 3, change the line NRSTPD = 11 back to 22 :-)
> 
> [MFRC522.py.txt](https://github.com/mxgxw/MFRC522-python/files/2896078/MFRC522.py.txt)

> Error is fixed but my tag not detected..

Same with me. I have writing functionality but now the written card is not being read by the same device... Any ideas?

alexball80, 2019-10-17:
Hi, if this can help, in my notes of a working test (project of some months ago) i've noted this:

(Estrema attenzione alle saldature e ai contatti - problema per giorni!)  <---- !!!!!!!!!
Translated: (Extreme attention to welds and contacts - problem during many days!) !!!!!! 
Make attention to welding.

I have the NRSTPD = 22 and this is my pin config:

RC522 Pin	Pi GPIO pin	PI GPIO signal and function
SDA		        24		        CS0 – This is the SPI chip select, active low
SCK		        23		        SCLK – this is the SPI clock
MOSI		19		        MOSI – data to the RC522
MISO		21		        MISO – data from the RC522
IRQ		          –	 	        -
GND		20		       GND
RST		        22		       GPIO25
3.3V		        17		       3.3V

daniellopez0708, 2019-11-14:
Hey guys

> Hi all,
> i stumbled on this as well on a RPI 3 B+ and Raspian Strech. First i changed GPIO 22 to 11, because of this comment "SPI_CLK on a RPi 3 is Pin number 23 = GPIO11 (not 22)" in [...]

Hey guys! So my issue is that the error is gone, but the tags are not detected... the pin config is identical to @alexball80 
Any ideas?

alexball80, 2019-11-14:
@daniellopez0708 
In my case one problem was welding contacts of shield
(Extreme attention to welds and contacts - problem during many days ,Make attention to welding)
Try with other RFID tags shields.

daniellopez0708, 2019-11-14:
@alexball80 
I will re-weld the contacts and will let you now

KhinShoon, 2019-11-15:
> Hi all,
> i stumbled on this as well on a RPI 3 B+ and Raspian Strech. First i changed GPIO 22 to 11, because of this comment "SPI_CLK on a RPi 3 is Pin number 23 = GPIO11 (not 22)" in [...]

Thank you for this quote. this solution works for me on python3 but not with python2.7

KhinShoon, 2019-11-15:
> Hey guys! So my issue is that the error is gone, but the tags are not detected... the pin config is identical to @alexball80
> Any ideas?

It was the same for me at first.
make sure your tag is close to the reader range and try moving the reader position from the breadboard. Sometimes, pins are not detected in sync. 
That is my problem and it works now.

vshashankv3, 2019-11-21:
> @te-ti emmm...I just want know have you fixed the problem?
> Change another one?Or fix the code?
> It's very important to me.
> Thank you very much!

First disable the SPI and then enable the SPI and then tyr again. Worked for me

vshashankv3, 2019-11-21:
"sudo raspi-config"
then go to interfacing option
then go to p4 SPI
disable that.
then "sudo reboot" // to reboot the pi
then again enable the spi interfacing by repeating above steps and reboot the system.

This thing worked for me..

daniellopez0708, 2019-11-22:
> > @te-ti emmm...I just want know have you fixed the problem?
> > Change another one?Or fix the code?
> > It's very important to me.
> > Thank you very much!
> 
> First disable the SPI and then enable the SPI and then tyr again. Worked for me

Hi! It is still not working for me but I will try what you said
Did you use the @alexball80 pin config?

wahajmurtaza, 2020-02-12:
Just found the solution that worked for me

git clone https://github.com/lthiery/SPI-Py.git
cd SPI-Py
git checkout -b 8cce26b9ee6e69eb041e9d5665944b88688fca68
sudo python setup.py install

After that **Enable Remote-GPIO** for raspberry pi interface.

 worked with python3 and 2 on Pi 4

CharanYvvs, 2022-03-09:
> git clone https://github.com/lthiery/SPI-Py.git
> git checkout 8cce26b9ee6e69eb041e9d5665944b88688fca68
> sudo python setup.py install

yes it worked for me. but when i place my rfid tag it is not detecting. can anyone help me out. its important for me as im using it for my major project .

wmx-meow, 2024-04-15:
> 英语很差,用中文回答吧.因为新提交的SPI-Py中,spi.transfer需要两个参数,在MFRC522.py中定义device_0=spi.openSPI(device="/dev/spidev0.0", mode=0, speed=1000000),修改 spi.transfer( ((addr<<1)&0x7E,val))和val = [...]

非常感谢这位大哥 解决了
this reply works and solved my problem.
a simple translation:
define the following line in the class MFRC522
'''
device_0=spi.openSPI(device="/dev/spidev0.0", mode=0, speed=1000000)
'''
then change these two lines:
'''
spi.transfer( ((addr<<1)&0x7E,val))
val = spi.transfer((((addr<<1)&0x7E) | 0x80,0))
'''
into:
'''
spi.transfer(self.device_0, ((addr<<1)&0x7E,val))
val = spi.transfer(self.device_0, (((addr<<1)&0x7E) | 0x80,0))
'''
简直在挖坟