mxgxw/MFRC522-python #75: GPIO.setwarnings(False) to disable

mhht, 2019-05-16:
Hello I have with the READ.PY probelm and do not know how to handle it. Help and thanks for the help.

/home/pi/MFRC522-python/MFRC522.py:133: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.NRSTPD, GPIO.OUT)
Traceback (most recent call last):
  File "read2.py", line 41, in <module>
    MIFAREReader = MFRC522.MFRC522()
  File "/home/pi/MFRC522-python/MFRC522.py", line 135, in __init__
    self.MFRC522_Init()
  File "/home/pi/MFRC522-python/MFRC522.py", line 406, in MFRC522_Init
    self.MFRC522_Reset();
  File "/home/pi/MFRC522-python/MFRC522.py", line 138, in MFRC522_Reset
    self.Write_MFRC522(self.CommandReg, self.PCD_RESETPHASE)
  File "/home/pi/MFRC522-python/MFRC522.py", line 141, in Write_MFRC522
    spi.transfer(((addr<<1)&0x7E,val))
TypeError: function takes exactly 2 arguments (1 given)

Comfubar, 2019-06-03:
Im also having around the Same issue with mine on a Raspberry Pi B+

Traceback (most recent call last):
  File "Read.py", line 41, in <module>
    MIFAREReader = MFRC522.MFRC522()
  File "/root/SPI-Py/MFRC522-python/MFRC522.py", line 135, in __init__
    self.MFRC522_Init()
  File "/root/SPI-Py/MFRC522-python/MFRC522.py", line 406, in MFRC522_Init
    self.MFRC522_Reset();
  File "/root/SPI-Py/MFRC522-python/MFRC522.py", line 138, in MFRC522_Reset
    self.Write_MFRC522(self.CommandReg, self.PCD_RESETPHASE)
  File "/root/SPI-Py/MFRC522-python/MFRC522.py", line 141, in Write_MFRC522
    spi.transfer(((addr<<1)&0x7E,val))
TypeError: function takes exactly 2 arguments (1 given)

Comfubar, 2019-06-03:
> 
> 
> Hello I have with the READ.PY probelm and do not know how to handle it. Help and thanks for the help.
> 
> /home/pi/MFRC522-python/MFRC522.py:133: RuntimeWarning: This channel is already in use, continuing anyway. Use GPIO.setwarnings(False) to disable warnings.
> GPIO.setup(self.NRSTPD, GPIO.OUT)
> Traceback (most recent call last):
> File "read2.py", line 41, in
> MIFAREReader = MFRC522.MFRC522()
> File "/home/pi/MFRC522-python/MFRC522.py", line 135, in **init**
> self.MFRC522_Init()
> File "/home/pi/MFRC522-python/MFRC522.py", line 406, in MFRC522_Init
> self.MFRC522_Reset();
> File "/home/pi/MFRC522-python/MFRC522.py", line 138, in MFRC522_Reset
> self.Write_MFRC522(self.CommandReg, self.PCD_RESETPHASE)
> File "/home/pi/MFRC522-python/MFRC522.py", line 141, in Write_MFRC522
> spi.transfer(((addr<<1)&0x7E,val))
> TypeError: function takes exactly 2 arguments (1 given)

So I found this on this issue rolled back like that one stated as well thought ehh its from 2016 Why not try 

Well now ive done everything It started the Read.py correctly but It just doesnt read them now 

https://github.com/mxgxw/MFRC522-python/issues/69

Did this all as well

I can now detect card on Read.py.
My problem was that I use python3 on Raspberry Pi3.
I do some change:

    after downloaded, sudo python3 setup.py install on /SPI-py (latest version)
    Then i've changed all print in MFRC522.py like this print "test" -> print("test")
    The NRSTPD is set to 11 in MFRC522.py

yamatopotter, 2019-06-07:
going back to python 2.7 don't solve the problem?