mxgxw/MFRC522-python #76: Error when trying to run read.py

lyocyhs, 2019-07-03:
File "Read.py", line 41, in <module>     MIFAREReader = MFRC522.MFRC522()   File "/home/pi/MFRC522-python/MFRC522.py", line 135, in __init__     self.MFRC522_Init()   File "/home/pi/MFRC522-python/MFRC522.py", line 406, in MFRC522_Init     self.MFRC522_Reset();   File "/home/pi/MFRC522-python/MFRC522.py", line 138, in MFRC522_Reset     self.Write_MFRC522(self.CommandReg, self.PCD_RESETPHASE)   File "/home/pi/MFRC522-python/MFRC522.py", line 141, in Write_MFRC522     spi.transfer(((addr<<1)&0x7E,val)) TypeError: function takes exactly 2 arguments (1 given)

Please help me to solve this problem

Freddy789, 2019-07-04:
look here i had the same issue and this helped me solve it 
btw look around before you open a question lol the link is the exact same problem you are facing
https://github.com/mxgxw/MFRC522-python/issues/69

lyocyhs, 2019-07-05:
I tried it, but I still get the same problem.
Sorry, but I also read the issue

SciMad, 2019-07-17:
Reverting the SPI-Py to 8cce26b resolved my issue as suggested in the following link.

https://github.com/mxgxw/MFRC522-python/issues/69#issuecomment-466023799

priyesh1112, 2019-09-04:
@lucassarcanjo Thanks a lot for your coordintation in making this post worth usable and improving the flaws in the syntax. I have followed the new directories which you have mentioned in the above comments. I am getting the below error right now. I would be glad if you could assist me,

Welcome to the MFRC522 data read example
Press Ctrl-C to stop.
Traceback (most recent call last):
  File "rfidreader.py", line 51, in <module>
    (status,TagType) = MIFAREReader.MFRC522_Request(MIFAREReader.PICC_REQIDL)
  File "/home/pi/MFRC522-python/MFRC522.py", line 239, in MFRC522_Request
    self.Write_MFRC522(self.BitFramingReg, 0x07)
TypeError: Write_MFRC522() takes 2 positional arguments but 3 were given

Thanks and Regards

victormagalhaess, 2019-09-05:
Hello @priyesh1112 , I think I got what can be the reason of your issue.
In @lucassarcanjo 's code, line 310, a variable called "backData" was misspelled as "ackdata", I send him a review for correcting his code, until it happens, you can change the line

```python
      print("Size: ".format(ackData[0]))
```
to

```python
      print("Size: ".format(backData[0]))
```

It should work.

priyesh1112, 2019-09-10:
@victormagalhaess thanks a lot for your feedback. It worked fine. Now I am facing some other issues which I am trying to rectify :)

priyesh1112, 2019-09-10:
/home/pi/MFRC522.py:133: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.NRSTPD, GPIO.OUT)
Welcome to the MFRC522 data read example
Press Ctrl-C to stop.
Traceback (most recent call last):
  File "/home/pi/rfidreader.py", line 51, in <module>
    (status,TagType) = MIFAREReader.MFRC522_Request(MIFAREReader.PICC_REQIDL)
  File "/home/pi/MFRC522.py", line 242, in MFRC522_Request
    (status, backData, backBits) = self.MFRC522_ToCard(self.PCD_TRANSCEIVE, TagType)
TypeError: 'NoneType' object is not iterable

victormagalhaess, 2019-09-10:
Hello @priyesh1112 , can you send me your code? I'll have an raspberry available tomorrow, so if you send me I can test and try to solve your issue with a practical approach (this way I can test if any other issues spots).

priyesh1112, 2019-09-11:
Thanks @victormagalhaess that is really kind of you. I am trying to solve it but if it does not happen, I will copy the code and send it to you so that you may verify. I have also consulted @lucassarcanjo but he has not responded yet. I will let you know. Thanks and Regards

priyesh1112, 2019-09-11:
@victormagalhaess could you share your email id so that I can send you my code? Though it is exactly the same as Lucas has posted. But, I am facing different error.