mxgxw/MFRC522-python #83: TypeError

Niwla23, 2019-09-28:
```
pi@raspberrypi:~/MFRC522-python $ python3 Write.py 
/home/pi/MFRC522-python/MFRC522.py:133: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.NRSTPD, GPIO.OUT)
Traceback (most recent call last):
  File "Write.py", line 41, in <module>
    MIFAREReader = MFRC522.MFRC522()
  File "/home/pi/MFRC522-python/MFRC522.py", line 135, in __init__
    self.MFRC522_Init()
  File "/home/pi/MFRC522-python/MFRC522.py", line 406, in MFRC522_Init
    self.MFRC522_Reset();
  File "/home/pi/MFRC522-python/MFRC522.py", line 138, in MFRC522_Reset
    self.Write_MFRC522(self.CommandReg, self.PCD_RESETPHASE)
  File "/home/pi/MFRC522-python/MFRC522.py", line 141, in Write_MFRC522
    spi.transfer(((addr<<1)&0x7E,val))
TypeError: function takes exactly 2 arguments (1 given)

```

I fixed the brackets but getting this error. please help

JoeLutz11, 2019-09-30:
i Have the same Problem!

gunalp, 2019-10-28:
i have

Daniel2193, 2019-11-18:
https://github.com/mxgxw/MFRC522-python/pull/80/commits/52507f18514bd4259c3e34886587ac2f95b13cb4

It fixes the errors but i cant get any readings -> card aren't detected but maybe someone can tell me why

Daniel2193, 2019-11-18:
I'll try it again tomorrow

Daniel2193, 2019-11-27:
I know a bit late but after you apply these patches everything works fine (27-11-19)

> 52507f1

> 

>

wahajmurtaza, 2020-02-12:
Just found the solution that worked for me

git clone https://github.com/lthiery/SPI-Py.git
cd SPI-Py
git checkout -b 8cce26b9ee6e69eb041e9d5665944b88688fca68
sudo python setup.py install

After that Enable Remote-GPIO for raspberry pi interface.

worked with python3 and 2 on Pi 4